mmiscool/esp8266Basic #10: Changing the listening port of the configuration webpage and adding htaccess-like authentication

crushedice2000, 2015-11-23:
These options would be great.

mmiscool, 2015-11-24:
Listening port is defined before program starts. Would be interested in a method of changing this. If you have a solution in mind let me know.
    ESP8266WebServer server(80);

Eszartek, 2015-11-24:
I would be interested to know the value of the webserver listening on a port other than 80.

The only use case I have would be to connect directly to my ISPs wifi network and perhaps be able to access the device from the internet (assuming they dont NAT the ip adresses) because they block port 80.

crushedice2000, 2015-11-24:
@Eszartek: What happens if do you have a ESP8266Basic device attached on your public wifi AP? This will add more difficulty level if any skillful person finds the ESP8266 IP address.

Also another reason to do this is that we could have the RUN tab on the port 80 and the rest of tabs (majorly configuration and/or debugging) on another port (i.e. 81)

Eszartek, 2015-11-24:
@crushedice2000: I suppose someone could perform a denial of service attack on my public facing ESP, since it only has 4 incoming socket connections available. 

Running the config on a separate port is probably possible using the Arduino libraries. Its just the Arduino libraries expect object declarations at the start of the app, not at runtime. Dynamic i/o config is not something thats typical for a typical Arduino device. I am looking for a solution.

crushedice2000, 2015-11-24:
@Eszartek: As I don't propose to change the listening port at runtime but in the configuration... What about storing the Run port and the Config port on the SPIFFS along with the essid, password and all the settings, and then recovering the port number and issuing `ESP8266WebServer server(LoadDataFromFile("RunPort"));`?

Eszartek, 2015-11-24:
Unfortunately, at that point, you are in "runtime" already. If it was easily doable, I would have done it. (I'd like to have the feature myself)

crushedice2000, 2015-11-24:
@Eszartek: Sorry, when I said *runtime* I meant *before starting the webserver*

I'll try to make a fork and try to implement it.

Eszartek, 2015-11-24:
You'll notice that the webserver object (with the port number) is created right at the beginning of the code, right after the header defines. At that point, the webserver is not started yet. Starting the webserver just starts it, it doesn't allow the configuration of the port number.

crushedice2000, 2015-11-24:
We could move that line below... Or put  `SPIFFS.begin();` above.

Eszartek, 2015-11-24:
Possibly, i'll try it.

crushedice2000, 2015-11-24:
However note that I'm not a C++ expert (nor intermediate level). So excuse me if I say any absurdity.

Eszartek, 2015-11-24:
It appears you'd have to modify the webserver/ethernet libraries. You can't run SPIFFS.begin() outside of the setup() function and you can't define server(port) inside a function.

As far as running 2 ports at the same time, I tried it and it didnt work. I read somewhere that you can have only one listening port, but I figured Id try anyway. The 2nd port shows "open" when I scan the ESP for open ports, but it doesn't respond. Maybe someone else will have better luck.

crushedice2000, 2015-11-25:
@Eszartek, you're right. However, at the SDK level, seems to be possible to open simultaneous TCP servers. This would be a issue/PR to the ESP8266Arduino webserver library.

Sorry. However, this needs further investigation.

theevilshiftkey, 2016-02-16:
I don't have an answer to this (my kung fu is too weak) but I do have a likely workaround. I've seen projects using the ESP with a generic TCP server that is started from the setup() on any port. It seems likely that one could use such a server to relay the alternate port traffic to the standard HTTP port internally. Food for thought.

crushedice2000, 2016-02-17:
@theevilshiftkey: Eating that food... :smile: Great!

Also the ESP8266Arduino team is doing a new async webserver. Maybe can be implemented port change on that new library.

mmiscool, 2016-05-07:
The port number has been added as a setting on the settings page.

closed 2016-05-07 by mmiscool

Joshfindit, 2017-11-01:
This seems like as good a place as any to ask: Is there a way to set up the ESP8266 so that requests to different ports can be interpreted as values?

Example: send a UDP request to port 1 = lowest LED brightness, Request to port 65535 = highest brightness.

More generally: can the server listen on a range of ports and pass the requesting port number to the code?

theevilshiftkey, 2017-11-02:
I'm pretty sure that this works:

`WiFiServer tcpserver(23);`

This would open said port, you'd have to handle it yourself in the loop. Having said that, it's MUCH better to run one server than 65535 servers (ports).