mmiscool/esp8266Basic #5: Could be replaced the use of the String class with char?

crushedice2000, 2015-11-23:
Hi! I've read [a post](http://hacking.majenko.co.uk/the-evils-of-arduino-strings) on a blog saying that the use of Arduino `String` class is a bad idea. Wouldn't be better to replace them by `char` and `char*`?

mmiscool, 2015-11-24:
This is not too big of a problem on the esp. I am not the best c/c++ coder so it might make it harder to maintain for my self.

Encapsulated in side functions should be fine how ever because those are limited in scope

crushedice2000, 2015-11-24:
Ok, I *never* used C/C++ I'm a mid-level Python programmer and a high-level Bash and Gambas 3 programmer. I never used C/C++ with a serious aim.

However you could say me which side functions could we convert...

closed 2015-12-20 by crushedice2000