mmiscool/esp8266Basic #7: The variable and function names aren't serious (or I believe so)

crushedice2000, 2015-11-23:
This is not a issue really (as in doesn't affect the code), but I think that should be fixed to improve readability.

Humorous examples (a selection of the best ones):

    GetRidOfurlCharacters
    WhatIwillReturn
    GetMeThatVar
    
Serious typos found:
    GraphicsEliments
    PinDesignaor
    VarialbeLookup

---

Dear maintainer:

  What about using global *Find and Replace* dialog of your favorite text editor to fix these things?

:stuck_out_tongue_winking_eye:

mmiscool, 2015-11-24:
A bit of humor. Also lack of spell check in code. Mabe I am a bit dyslexic

crushedice2000, 2015-11-24:
I have to recognize that they are humorous :smile: (even for non-native English speakers -as me-). However I think that this could be a serious pain for future contributors (even me).

I think you should put the humorous part in the comments and not in the code (this is my opinion, however, you're the project owner). :stuck_out_tongue:

closed 2015-12-20 by crushedice2000