mmiscool/esp8266Basic #8: Code should be cleaner (and I would help to clean it up)

crushedice2000, 2015-11-23:
I think that the code should be cleaner and also observe these rules:

1. We should put a list of dependencies (Libraries).
2. The code should be more structured.
3. We could optimize the functions to be smarter.

mmiscool, 2015-11-24:
1. A dependency list should be generated. Just have not gotten around to this yet. Would need a list with links to download each.

2. what kind of structure are you thinking?

3 Optimization is always good and welcomed.

crushedice2000, 2015-11-24:
1. I'll do later (however I'm facing problems to compile on 1.6)

2. This proposal is directly linked to the issue #7, also I refer to put <s>more</s> meaningful <s>yet</s> names to the project files

3. I'll try to review your code, however, will be hard. You have dyslexia and I have dyscalculia. :smiling_imp:

crushedice2000, 2015-11-24:
4. The HTML code is ugly. Wouldn't be better to store the html files as files on the spiffs and then replace parts with String.repalce?

crushedice2000, 2015-11-24:
5. I kept reading your code and it's like the output of a decompiler :smiling_imp:

crushedice2000, 2015-11-24:
`6.` Can you document the use of 9 blank lines form line 68 to line 76 on ESP8266Basic.ino? :smiling_imp:

Eszartek, 2015-11-24:
re: #4, how do you propose getting the data into a formated spiffs file? (it can't be put into the firmware image as far as I can tell)

Eszartek, 2015-11-25:
My point of view (just a different one, not meant to be combative):
1. How do you know when its cleaner, or clean enough? 
2. It appears to be structured so that it works.
3. Like #1, how do you know when is it smarter, or smart enough? 
4. HTML code is HTML code, it is what it is, I see no problem with it.
5. I can read the code fine, its amazingly clear.
6. It has no negative effect except on your eyes :)

The authors coding style is no better or worse that anyone else's. It so happens that it is similar to my own.

closed 2015-12-20 by crushedice2000