mmiscool/esp8266Basic #9: Styling the web interface

crushedice2000, 2015-11-23:
What about adding a small CSS to the web interface to make it look better?

mmiscool, 2015-11-24:
Could be nice. A bit concerned about binary size increases but would be fine to do. 

The basic interface today is rather bare bones and simple. Works on laptops. Tables and phones. Want to make sure it stays simple.

crushedice2000, 2015-11-24:
We could do a minimized CSS and then put the *big and commented* version here on GitHub... So the size is not a major deal (or I think so).

I respect the simplicity, but a bit of style would be great :smile:

mmiscool, 2016-04-18:
this has been done.

Upload a file named skin.css using the file manager. Interface can now be skinned.

closed 2016-04-18 by mmiscool