mmiscool/esp8266Basic #12: Missing hader file when compiling

ztz472947849, 2015-11-24:
#include "spiffs/spiffs.h"
#include <FS.h>
#include <ESP8266mDNS.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
//#include <WiFiUdp.h>
#include <ESP8266WebServer.h>
#include <Wire.h>
#include <ESP8266httpUpdate.h>
#include <Servo.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include "Base64.h"

#include <WiFiClientSecure.h>
#include "ESP8266httpUpdate.h"
#include <time.h>

//LCD Stuff
#include <LiquidCrystal_SR.h>
#include <I2CIO.h>
#include <LiquidCrystal.h>
#include <LiquidCrystal_SR3W.h>
#include <LCD.h>
#include <LiquidCrystal_SR2W.h>
#include <FastIO.h>
#include <LiquidCrystal_I2C.h>

//PS2 Key Board
#include <PS2Keyboard.h>

i have only a few of these .h files after installing the ArduinoIDE + esp8266 borads patch in a json format file.

crushedice2000, 2015-11-25:
What means all these empty includes?

Could you edit the issue title to replace *hader* by *header*?

Also, please put the code as code (appending 4 spaces to each line).

https://guides.github.com/features/mastering-markdown/

Also note that we haven't a dependency list yet. Sorry. We are working on it.

mmiscool, 2015-11-26:
Libraries I use currently on the build machine are in the libraries folder part of this git.

closed 2015-11-26 by mmiscool