mmiscool/esp8266Basic #13: Documentation in GitHub

riban-bw, 2015-12-20:
It would be advantageous to be able to download the documentation. It would further be advantageous to be able to contribute updates / corrections to the documentation. These two may be facilitated by hosting the documentation within a github repository, maybe the same repository as the source code.
When a user configures the ESP8266 as an access point and connects to it to program in BASIC, the user may lose Internet connection which in turn loses access to documentation, hence the advantage of an off-line copy.
It may be advantageous to offer the ability to bundle the documentation in to the ESP8266 BASIC web interface but this may use up too much device storage.

todocono, 2016-03-31:
I have to second riban-bw. I am looking at this project to do something serious about it. Right now it is difficult to convince anybody to use this for educational purposes if the docs are static in a website. Please check readthedocs.org to connect it with github. I am making a pull to start playing around. This is a terrific project !!

mmiscool, 2016-05-05:
documentation is now available in a google doc. It can be downloaded as a PDF.

closed 2016-05-05 by mmiscool