mmiscool/esp8266Basic #14: Cannot clear corrupt BASIC code

riban-bw, 2015-12-20:
I have saved a BASIC code script which causes the ESP8266 to become unresponsive. (This may be a tight loop or bad syntax - code was incomplete when I accidentally hit save.) ESP now starts up, runs code and becomes unresponsive on web interface. Reflashing BASIC interpretor does not fix, presumably because this does not clear the BASIC program memory so the bad code persists.
It may be advantageous to clear program memory when flashing device, or at least offer option to do so.

mmiscool, 2015-12-20:
Upon boot of the device you have 30 seconds to save a blank program. This should fix the condition your esp is in.

riban-bw, 2015-12-21:
Thanks - I was able to do that (though it was tricky, waiting for the ESP access point to become available, connecting to it, opening (refreshing) the webpage, and saving a blank script within 30 seconds). I now think that I did not have a corrupt script but did not understand the startup and programming process. I can't this documented anywhere. I am assuming that, if there is a default script then it is run after 30 seconds and that the user (programming) interface is not available when a BASIC program is running. Will you please confirm those assumptions (or correct them) and add this detail to the documentation?

I may find over the next few days to review the source code which may give me answers I seek but it makes sense to have an overview within the documentation that explains at a high level how the system works.

Good work on this. I am glad to see others appreciate that BASIC is a useful tool and not just a learning tool from the 1980s!

Cheers

closed 2015-12-21 by riban-bw