mmiscool/esp8266Basic #16: File I/O failed

Derek-K, 2015-12-30:
Just installed version 1.68 (Tired both 1MB and 512k version on ESP8266-1 with 1MB flash)

I was able to connect to it via WiFi (SSID=ESP) and browser able to access 192.168.4.1. However I was not able to save config. Tried to hit "Format" button, it just print "1", tried to save but getting "File Open Failed"

I have also tried to use "EDIT" then save, I get the same error too. Looks like File System access is broken at the moment?

I have putty connected at 9600 BAUD and these are the output.

SP Basic 1.68
1
2
3
4
5
6
7
8
9
10
11
12

Failed Wifi Connect
Creating WIFI access point
ESP
APname
file open failed
APpass
file open failed
Starting Default Program
Done...
WIFIname                <-- trying to save config
file open failed
WIFIpass
file open failed
APname
file open failed
APpass
file open failed
LoginKey
file open failed
ShowMenueBar
file open failed
otaUrl
file open failed
Formating   <-- Trying to "Format"
1Formating   <-- and again
1WIFIname
file open failed
WIFIpass
file open failed
APname
file open failed
APpass
file open failed
LoginKey
file open failed
ShowMenueBar
file open failed
.......
Done...
default/1   <--- trying to save the text in "Edit" window
file open failed
default/2
file open failed
default/3
file open failed

On a side note. For some interesting reasons, when I use ESP_Basic_Flasher.exe to flash, my ESP8266 seems unstable and will crash randomly and reboot. But if I flash ESP8266Basic.cpp.bin manually using ESP8266Flasher.exe seems more stable (no crash yet, other than first boot)

todocono, 2016-03-31:
Any follow up on this? @Derek-K ? did it keep happening?

Derek-K, 2016-03-31:
Sorry to tell you I have abandon using ESP Basic shortly after that....

todocono, 2016-03-31:
I have been tinkering all day around it and still didn't get to see many inconsistencies. May be this issue was already fixed. Thanks for the input.. too unstable to work on or did you just got busy with other projects?

Derek-K, 2016-03-31:
Sadly both. Too unstable to work on and moved on...

It was a good project, but still liking NodeMCU project more...

todocono, 2016-03-31:
I don't know exactly what did I do but I ended up with the same I/O error you have ... working on it but it looks like I flashed the wrong version or something like that 

@mmiscool : can you throw some light on what to check? I am on a nodemcu dev board v0.9 , any idea which firmware should I be flashing and if that is the right approach to tackle the file writing problem?

mmiscool, 2016-07-09:
I don't think this is a problem after moving to newer esp arduino package.

closed 2016-07-09 by mmiscool