mmiscool/esp8266Basic #20: Crash after compiling in own environment

todocono, 2016-03-31:
I am trying to reproduce the compiling process. My setup is:

Arduino 1.6.5 (previous experience with ESPEasy taught me to stick to this version, please advice if this was a wrong supposition)
ESP82666 v2.0 
Board ESP Node MCU v0.9
Installed all libraries included in the current repository

It does compile. But when starting, it shows on the terminal that it is logged in into the local LAN. Address is something like 192.168.10.104. By either accessing this or accessing the 192.168.4.1, the system makes a dump and reboots itself. 

Please advice

todocono, 2016-04-01:
I tried to switch to Arduino 1.6.8 but no luck at all. This is what I see on the serial:

Connected to 
IP address : 192.168.10.104
WIFIname
WIFIpass
file open failed
Starting Default Program
Done...

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x402535f3 depc=0x00000000

ctx: cont 
sp: 3fff4440 end: 3fff4720 offset: 01a0

>>>stack>>>
3fff45e0:  40107270 00000001 3fff8768 0000001f  
3fff45f0:  402535f3 3fff8738 3fff7da8 402212e8  
(continues with the stack)

Any chance that what I need is to format / setup the module before flashing it ? Some previous condition that it's not letting it go through the normal process of loading it's configuration file?

mmiscool, 2016-05-05:
It looks like it is safe to use the latest stable esp8266 arduino environment,

closed 2016-05-05 by mmiscool