mmiscool/esp8266Basic #26: msgreturn always returns "No MSG Branch Defined"

RobertoNagy, 2016-08-04:
Running the MSG URL explains the msgreturn always returns "No MSG Branch Defined"

mmiscool, 2016-08-05:
This should be corrected in current version. If it is not reply to this with a small example program so I can test and get rid of the bug.

RobertoNagy, 2016-08-06:
Using this example:

msgbranch [mybranch]
print "set the branch"
wait

[mybranch]
myColorVar = msgget("color") 
print myColorVar
let myReturnMsg = "You Entered " & myColorVar
msgreturn myReturnMsg
wait

and using: wget http://192.168.4.1/msg?color=blue -q -O -
or in a web browser: http://192.168.4.1/msg?color=blue
the response is:  No MSG Branch Defined
The right response "blue" pops up at http://192.168.4.1/input? address.

mmiscool, 2016-08-11:
Fixed in latest web sockets branch. New build tonight

closed 2016-08-11 by mmiscool

gdelconti, 2018-11-03:
Hello , this bug is still present under ESP Basic 3.0.Alpha 69.
Simply add  one "delay x" statement to the above example, like this:

msgbranch [mybranch]
print "set the branch"
wait

[mybranch]
myColorVar = msgget("color")
print myColorVar
delay 10  ' JUST ADD THIS LINE
let myReturnMsg = "You Entered " & myColorVar
msgreturn myReturnMsg
wait