mmiscool/esp8266Basic #32: Bug: IDE Editor Open Appends Blank Line

flywire, 2017-01-29:
Go to IDE
Select a file in File Manager and click File Manager Edit button or just go to Editor and enter a filename then click Editor Open button
Go to end of file and you will notice that a blank line has been appended.
Click Save, Click Open and go to end of file and another blank line has been appended. This occurs repeatedly.

This bug is repeatable.

mmiscool, 2017-02-23:
The blank line being added prevents a single line program from crashing the system.
It seems this is not such a big problem to me.

flywire, 2017-02-23:
I agree that it is not critical. An editor that changes the program every time it is read? Can't that instance be trapped?