mmiscool/esp8266Basic #35: No [VARS], no [DEBUG] in menu

Kabron287, 2017-06-07:
ESP Basic 3.0.Alpha 69, 8MBit Wemos D1 mini lite. Everything works, but no [VARS], no [DEBUG] in menu.
[image]
Any Ideas?

themotes, 2018-01-14:
I have the same problem with the esp8266 01. I flashed version 3 with 1M, also 512, etc. What can we do?

mmiscool, 2018-01-14:
I believe I made the builds for those size modules omit these pages to make it fit in the limited space.

themotes, 2018-01-15:
Thanks. Any suggestions for debugging with these modules?

mmiscool, 2018-01-15:
Most of the time I would do my debugging on an esp-12 which has 4 megs of flash. 
Then I would switch over to the smaller modules. 

I am surprised that there are still venders selling the older modules as the esp-12 modules can be had for just as cheap.

josmaroal, 2018-12-29:
Translation:
Hello Michael, start testing your ESPBasic program at the end of October 2018, surpassing the language (I speak Spanish) it was amazing the progress in making an ESP-12 work, it was a return to what I was able to know in the 70's! IT WAS EXTRAORDINARY!
It was my best birthday present (I turned 70 on 11/11/2018)
Thanks for that gift, friend
MERRY CHRISTMAS AND A LOT OF PROSPERITY THE NEXT YEAR
Note: I would like to talk and be able to ask some questions, do you always enter?